DIY-Machines/SmartChess #2: Can not compile arduino code

olskar, 2021-04-25:
Hello!

Trying to compile the arduino code Chess_Simp.ino

`C:\Users\Åskar Andersson\Documents\Arduino\chess\chess.ino: In function 'void setup()':
chess:80:3: error: 'showChessboardMarkings' was not declared in this scope
C:\Users\Åskar Andersson\Documents\Arduino\chess\chess.ino: In function 'void loop()':
chess:88:3: error: 'lightUpMove' was not declared in this scope
chess:100:5: error: 'showChessboardMarkings' was not declared in this scope
C:\Users\Åskar Andersson\Documents\Arduino\chess\chess.ino: In function 'String humansGo()':
chess:137:5: error: 'lightUpMove' was not declared in this scope
chess:146:7: error: 'showChessboardMarkings' was not declared in this scope
C:\Users\Åskar Andersson\Documents\Arduino\chess\chess.ino: In function 'void waitForPiToStart()':
chess:223:5: error: 'showChessboardOpeningMarkings' was not declared in this scope
chess:226:25: error: 'loadingStatus' was not declared in this scope
C:\Users\Åskar Andersson\Documents\Arduino\chess\chess.ino: In function 'void errorFromPi()':
chess:311:3: error: 'showChessboardMarkings' was not declared in this scope
C:\Users\Åskar Andersson\Documents\Arduino\chess\chess.ino: In function 'void hint()':
chess:526:20: error: 'loadingStatus' was not declared in this scope
chess:530:5: error: 'showChessboardMarkings' was not declared in this scope
chess:539:5: error: 'lightUpMove' was not declared in this scope
chess:540:5: error: 'showChessboardMarkings' was not declared in this scope
C:\Users\Åskar Andersson\Documents\Arduino\chess\chess.ino: In function 'void checkForComputerCheckMate(String, String)':
chess:585:7: error: 'showChessboardMarkings' was not declared in this scope
exit status 1
'showChessboardMarkings' was not declared in this scope

Any ideas?
`

olskar, 2021-04-26:
Missed that Chess_Simp.ino and chessboard_LEDs.ino need to be compiled together, closing :)

closed 2021-04-26 by olskar

DIY-Machines, 2021-04-26:
Thanks for closing this one - I was just going to write out the same response. :)

lhops9331, 2021-12-05:
Lewis,
I see that someone was having the same problem as I but he cured it by combining the two sketches.
I can't seem to find how to combined sketches .  Can you please guide me to the instructions. the text below
is from an email I sent you yesterday. I am hoping that you answer this email quickly...I am dying to get this 
board up and running.

I am having problems with compiling two of the sketches using Arduino IDE.  I continue getting errors
“Xxxxxx not in the scope”.  I was able to compile and upload  electronic_test to the Nano without problems.
But chess_simp and chessboard_LEDs …not having any luck.  Do these sketches have to be combined
And if so how is that done.  Very new to the IDE. I had no problem with the Raspberry Pi or any of the wiring 
Everything seems to be working as far as the LEDs with the electronic_test loaded.  I have all the parts printed 
And hoping you can help.  Looking forward to spending time with this chess board. 

Thank you for your help.

Leon
hoppy@hoppy1.net