DIY-Machines/SmartChess #3: Crashing on move entry

hermy65, 2021-06-11:
We built one of the chessboards for my son to play and we cant even get through a full game without it crashing. Sometimes he will make it 1/2-3/4 of the way through a game, enter a move then it starts doing the green cell by cell illumination and the screen will still say your move but then it wont accept any new moves. Sometimes if he presses the hint button the text turns blue then  then coordinates turn multi colored then the green cell thing happens again. So far he has tried to play games 7 times and can never complete one because the green cell thing happens. 

Running on all of the hardware in your build guide except we are using a pi 3 b instead of the a

cheepiao, 2021-06-15:
I am building one too. Now, I am waiting for some of the electronics parts to arrive. Anyway, I will try it out once I completed it. 
Will update here. Thank you for sharing.

cheepiao, 2021-06-21:
> We built one of the chessboards for my son to play and we cant even get through a full game without it crashing. Sometimes the crash just happens randomly and the board does the startup green cell by cell animation mid-game when the screen says thinking, other times, and most often its when he pushes the hint button. Sometimes when using the hint button the coordinates change to multicolors and then the board crashes, other times it just hangs there and does nothing. Below are 2 videos, one showing using the hint, then the coordinates going multi colored then crashing and one showing the hint button just causing the board to hang and never do anything.
> 
> [Video 1 - multi colored coordinates and a crash ](https://youtu.be/5zU6Wc5abH8)
> [Video 2 - hint button leading to board freezing, bad coordinates in the screen and nothing happening on the board itself](https://youtu.be/1PeP0Ac6_wM)
> 
> Running on all of the hardware in your build guide except we are using a pi 3 b instead of the a

I just wondering if the power supply plays a part? The NeoPixel LED should takes up a bit of power from the power supply and it leave very little for the Raspberry Pi and others if the power supply is under powered. Thus, unstable power supply may cause strange issue. Well, just a thought, not sure if this could be the real cause. Hope you could find the root cause so that your son could enjoy the game soon. Cheers! :)

DIY-Machines, 2021-06-21:
Insufficient power (as @cheepiao suggests) or perhaps a shorting wire could be likely candidates. What is the power supply you are using at the moment?

hermy65, 2021-06-21:
Using a CanaKit Raspberry Pi 3+ power supply right now. Is there one you would recommend instead?

DIY-Machines, 2021-06-21:
That should be plenty of power available. To be certain let's dial down the brightness of the playing surface LEDs. Cn you re-upload the code to your Arduino Nano but change line 74 of 'Chess_Simp.ino' 

>   chessboardLEDS.setBrightness(255);
  
Change the value of he brightness from 255 down to something like 100. Does the issue still present itself?

I know it's a little awkward, but can you play it with the control panel (the one with th buttons and the one with the screen) unscrewed so that you can have internal electronics spaced out as much as possible whilst you play to also see if it's a wire shorting a circuit and causing a reset.

cheepiao, 2021-06-21:
Hi Lewis, glad that you share the 'Chess_Simp.ino' code to lower down the brightness. I was about to ask how to do there. Appreciated and thank you.

cheepiao, 2021-06-22:
> @DIY-Machines
> 
> Using a CanaKit Raspberry Pi 3+ power supply right now. Is there one you would recommend instead?
> 
> This one to be exact - https://www.amazon.com/CanaKit-Raspberry-Supply-Adapter-Listed/dp/B00MARDJZ4/ref=sr_1_3?crid=33UO4M985DU2M&dchild=1&keywords=micro+usb+power+supply&qid=1624291982&sprefix=microusb+power%2Caps%2C210&sr=8-3

Ok.... however, based on the rating by the users in Amazon, it does not seem very promising in keeping stable voltage upon higher load. Anyway, I had tried to reduce the brightness for all LEDs (both chess board and panel) from 255 to 20 and the power consumption really reduce pretty significantly. I did not do any electrical measurement but I just touch the power adapter and it feel much cooler. Hope this helps. Have fun. Cheers!

hermy65, 2021-06-22:
@DIY-Machines he played 2games with all of the electronics pulled out, he said he got to about 2 moves left in the game and it started the green cell thing each time. We made the brightness change and the same thing happened.

hermy65, 2021-06-25:
@cheepiao have you found any other power supply options worth looking at?

cheepiao, 2021-06-25:
@hermy65, actually I ordered a PWR+ power supply from Amazon before I tuned down the brightness of the LEDs. Anyway, the power supply is on it way and will try it out once it arrive. Have a wonderful weekend.

cheepiao, 2021-06-25:
@DIY-Machines, 
Actually, I would like to highlight something about the experience while playing so far... I never get to complete a game and when I get to "Checkmate" the computer, it always stay at the "thinking" state and never show any "end game animation" that I have won. Also, it seems I only can activate the "shutdown" sequence (by holding the Hint and H8) at the end of the game but not during game play. Anyway, I will try again when my new power supply arrive and see how it goes... Thank you.

DIY-Machines, 2021-06-28:
@cheepiao 
I believe you may have found a bug in the code. Do you mind creating a new 'issue' on Github for Checkmate handling and repeating your information. I'm rebuilding my chessboards so I can re-run and check the code for the issues you are finding. The end game ones were particular hard to troubleshoot as I know it was working when I added the end game handling but it seems something else has 'broken' that bit of code and I didn't always have the time to play lot's of games whilst refining the software - particular right to the end. :/

cheepiao, 2021-06-29:
@DIY-Machines
I had created a new issue entitled "Checkmate handling and shutdown ". Really appreciated it. Thank you.

DIY-Machines, 2021-07-08:
Just an update: I'll have finished putting my chessboard back together today. I have decided to first figure out a way to log what's happening so that when the problem surfaces itself you can send a log for me and others to look at to help accelerate  troubleshooting. Once this logging is added to the software I'll provide a link to the revised code which will need applying to your board(s).

DIY-Machines, 2021-07-11:
@hermy65  I've got to join the family on a holiday this week and will be scolded if they find me working on this. :)  For the time being can I ask if you mind trying to capture one or two examples of your board crashing out of a game? If you don't mind, please do the following:

- Power up your chessboard
- once the start up sequence is complete VNC to your Raspberry Pi (or SSH or connect a screen and keyboard)
- Open a terminal (if you VNC'd) and then navigate to SmartChess/RaspberryPiCode/   " cd SmartChess/RaspberryPiCode/ "
- Start the chess game script with "python2 StartChessGame.py"
- Press CTRL + C to interrupt the code and then launch it again using "python2 StartChessGame.py"
- Press the reset button onto of the Arduino Nano

You should then be able to play a game whilst monitoring the output from the Raspberry Pis code in the terminal. When the game crashes please can you copy the output from the terminal and post it here along with a description of your last few physical interactions with board?

hermy65, 2021-07-12:
Doing the same here too ;)

Will give it a shot when we get back home

hermy65, 2021-07-20:
@DIY-Machines tried it but i dont think its working as intended. I started the game manually like stated, did a control+c, started it again then pressed the reset button on the arduino. Once pushing the reset button on the arduino it went back to the green square loading animation and nothing else ever happened. This is what putty looked like:

`pi@raspberrypi:~ $ cd SmartChess/RaspberryPiCode/
pi@raspberrypi:~/SmartChess/RaspberryPiCode $ python2 StartChessGame.py

 Sent to board: heyArduinoChooseMode
Waiting for mode of play to be decided on the Arduino

 Waiting for command from the Board
^CTraceback (most recent call last):
  File "StartChessGame.py", line 297, in <module>
    gameplayMode = getboard()[1:].lower()
  File "StartChessGame.py", line 88, in getboard
    if ser.in_waiting > 0:
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 468, in in_waiting
    return struct.unpack('I', s)[0]
KeyboardInterrupt
pi@raspberrypi:~/SmartChess/RaspberryPiCode $ python2 StartChessGame.py

 Sent to board: heyArduinoChooseMode
Waiting for mode of play to be decided on the Arduino

 Waiting for command from the Board
Traceback (most recent call last):
  File "StartChessGame.py", line 297, in <module>
    gameplayMode = getboard()[1:].lower()
  File "StartChessGame.py", line 89, in getboard
    btxt = ser.readline().decode('utf-8').rstrip().lower()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
pi@raspberrypi:~/SmartChess/RaspberryPiCode $
`

DIY-Machines, 2021-07-26:
What appears to be happening is whilst the Pi is trying to communicate via the serial connection with the Arduino something is preventing a stable serial connection from being established/maintained. This is proving tricky to troubleshoot - sorry. 

- We've tried it with the electronics all spread out so that we know  - this was to check that the act of pushing buttons on the panel was not causing any connection to short between the wires, buttons and control boards.
- We know you have had some near complete games so it highly unlikely the software setup is incorrect as you got that far.
- For the same reason as above we can expect the hardware to be correct.

Intermittent bugs are the worst to identify and solve. Let's see if we can be 100% sure of the software - is their any chance you're in the UK? Then I can post you a loaner SD card which I know works on my chessboards.

Next will be bypassing the LLC and connecting the Raspberry Pi and Arduino with a USB cable to see if that makes the connection more stable. (This is not like this in the original design because of space constraints in the enclosure).

hermy65, 2021-07-26:
@DIY-Machines Unfortunately im not in the UK. If you have a sd card you want to take an image of and upload it somewhere i can try that. 

As for connecting the pi to the arduino, anything i need to change code wise to make that work?

DIY-Machines, 2021-07-26:
I think (from memory) we just need to update the port which the Arduino is connected on. See line #15 of StartChessGame.py. You may need to plug and unplug the USB whilst issuing `lsusb -t`

Uploading an image was my preferred method when creating the instructional video but the file was going to be several GB in size. I'll send you a private message with a link to download a copy of my image shortly when it has finished uploading. (It will still have my private adafruit.io credentials in it so I would appreciate it if you changed them if you keep using this image - thanks.)

DIY-Machines, 2021-07-26:
@hermy65 Sent image file to you via Patreon.

DIY-Machines, 2021-07-27:
@hermy65 I would test the image first as using the USB cable will involve physically disconnecting the serial wires from the Pi and Nano.

hermy65, 2021-07-27:
@DIY-Machines Maybe i did this wrong but i updated line 15 to reflect ttyUSB0 vs ttyAMA0 and on the first couple of attempts to get things working it would do the green loading animation, go about half way through then restart it all over again and never connect.

secondly, i tried your image and it never connects just slowly goes through the green loading animation and never starts up

DIY-Machines, 2021-07-28:
OK thanks for trying this. Just to clarify:

You tried the new image I sent on your own chessboard without any hardware modifications which is when it does the green loading and never starts up?

Then you tried to change the serial connection used by updating line 15, disconnecting the LLC at all points (and probably you should also disconnect the 5v power wire on the Nano as it will dry power from the Pi's USB) and then used a USB cable instead between the Pi and Nano. This then caused it to repeatedly loop the green loading screen animation?

Thanks

hermy65, 2021-07-28:
@DIY-Machines  Correct on the image, no modifications and it failed to start up.

As for the other statement, i did NOT disconnect the LLC. It appears that i incorrectly assumed i just needed to update line 15 and then test it. I can disconnect it as you asked above and see what happens

hermy65, 2021-07-30:
@DIY-Machines removed the LLC and the 5v wire, updated line 15 then plugged in USB and it started up fine. My son got maybe 8-9 moves into a game before things went sideways though. He attempted to move some piece from c2->c3 and the screen showed c0->c3 THINKING and then never progressed from there.

DIY-Machines, 2021-08-16:
Hi @hermy65 . Sorry it has taken sometime for me to get another look at this. Someone else solved and issue by ensuring they had the python-serial library installed. Please can you try this as see if it helps?

sudo apt install python-serial

As you are making some moves this might not be your exact problem but it will be worth ago. The other people who have done this have fully functioning boards again.

hermy65, 2021-08-17:
@DIY-Machines it appears that everything is working now going the USB route. Would you suggest we hook the LLC back up and then install the serial library and go that route or since its working now just leave it as is?

DIY-Machines, 2021-08-17:
If it is working for you with USB then I would suggest you keep enjoying your board as it is. If you're happy we can mark this as closed but it will still be here for other peoples reference.