DIY-Machines/SmartChess #8: No keyboard after start-up

KixMan28, 2021-07-17:
Hi Lewis! Today I finally finished my chessboard. After I plug in my board, the chessboard is lit up and the green boxes start to roll. They reach the end of the 4th row and the OLED displays "Choose opponent: 1.... 2 ....", but the rest of the green boxes won't lit up at once , but they continue to lit up sequentially. When they reach the end of the board, the keyboard won't lit up. If I press any number button, nothing happens. If I press the HINT button the green boxes disappear and the keyboard lights and the HINT are on. If I press any number button nothing happens, The display won't change. 
If I press again the HINT button, it will turn blue and the keyboard lights are off. If I press again the HINT, the keyboard lights are ON again, but still nothing happens if I click them.

Any idea what is going wrong?

DIY-Machines, 2021-07-19:
Hi @KixMan28 . This sounds like a problem with the Serial connection between the Raspberry Pi and Arduino Nano. I would start by carefully double checking your wiring as it goes into and out of the shift register. 
Also please check you completed the software steps to prepare the serial connecon on the Pi as well. Briefly they are:

`sudo nano /boot/config.txt`
Add the line: `dtoverlay=disable-bt`

`sudo nano /boot/cmdline.txt`
Find the following text and remove it:
`console=serial0,115200`
Save and exit then shutdown the Pi before power cycling the entire chess board.

KixMan28, 2021-07-19:
Hi @DIY-Machines ! Thank you for your reply. I have checked the wires from the Arduino Nano to the LLS and the wires from the RPi to the LLS. All wires are OK and provide good connection. I have also checked the voltages: On the B side I have 5V on the A side I have 3.3V. If I check B1 with A1, B2 with A2, etc should I have between them continuity or not? I have checked it on my module with or without power applied and there is NO continuity between the two sides. Does this mean my TXS0108S module is broken?

DIY-Machines, 2021-08-16:
Don't quote me on this, but I don't think a continuity test on a multi meter works though a LLS. Some people had a similar problem found installing the python-serial library helped:

sudo apt install python-serial

Please could you try this and let me know if it solve your problem?